frederik-h/acer-wmi-battery #2: Difference from another repo?

Brli, 2022-08-03:
namely, this one: https://github.com/maxco2/acer-battery-wmi

frederik-h, 2022-08-18:
I was not aware of this repository when I implemented my driver.   
This driver seems to implement essentially the same functionality for enabling the battery health mode, but I don't see an option to enable the battery calibration.

closed 2022-08-18 by frederik-h

agusti30, 2025-05-02:
I got this warning, sorry I don't know how to switch gcc version or install multiple gcc version in Fedora. Try to look it up in Internet still didn't work. 

warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250418 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)