frederik-h/acer-wmi-battery #6: Skipping BTF generation

tcaceresm, 2022-09-15:
Hi, I'm having this problem
`make -C /lib/modules/5.15.0-47-generic/build M=/home/tcaceres/acer-wmi-battery modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
  CC [M]  /home/tcaceres/acer-wmi-battery/acer-wmi-battery.o
  MODPOST /home/tcaceres/acer-wmi-battery/Module.symvers
  CC [M]  /home/tcaceres/acer-wmi-battery/acer-wmi-battery.mod.o
  LD [M]  /home/tcaceres/acer-wmi-battery/acer-wmi-battery.ko
  BTF [M] /home/tcaceres/acer-wmi-battery/acer-wmi-battery.ko
Skipping BTF generation for /home/tcaceres/acer-wmi-battery/acer-wmi-battery.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
`
what should i do

frederik-h, 2022-09-27:
This looks like a generic compilation issue to me. Have you built other kernel modules on this system? This [discussion](https://access.redhat.com/discussions/6970514) suggests that this is just a warning and that the module could actually have been built. Perhaps you can ignore the warning.

closed 2022-09-27 by frederik-h

mochechan, 2024-03-12:
Install the following package before making.
```
sudo apt -y install dwarves
sudo cp -avi /sys/kernel/btf/vmlinux /usr/lib/modules/`uname -r`/build/
```