frederik-h/acer-wmi-battery #7: Feedback: 12th gen Aspire 3

R3G3N3R4T0R, 2022-09-23:
# TL;DR
Health mode and Calibration mode work as expected and largely consistent with previous reports. Calibration effect untestable.
# System Environment
- OS: Artix Linux (kernel: 5.19.8-artix1-1)
- Host: Aspire 3 A315-59-53C5
# Details
- Standard features work as expected, Health mode stops charging above 80% while Calibration mode alternate between charge and discharge cycle.
- I am not able confirm whether the calibration mode really does anything, I tried shutting down every app and turn on calibration mode for >6 hrs at 80% initial charge. No change at `/sys/class/power_supply/BAT1/charge_full` even after reboot.
- Health mode does not release battery charge >80%, as if all the charge is going to the laptop and the battery is separated.
- `cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` does not always report 0, but it is unreliable regardless, it probably keeps its value when it is loaded.
- Battery status is changed to `Not Charging` when Health mode is triggered, It may take time to recalculate charging time when Health mode is disabled. Rest of the system still know it is charging.

I also wrote a [small script](https://gist.github.com/R3G3N3R4T0R/efc2e84e17b3acf8c6ba37cd002874db) to help a bit with all the memorizing. Kernel module still needs to be manually compiled every kernel update.

frederik-h, 2022-09-27:
Thank you for the feedback!

* When I enabled the calibration mode on my laptop, it started discharging immediately - even with the attached power supply.
After recharging the battery run time estimate became much more accurate (I had never calibrated the battery before).    
Could you check if the module reports that the calibration mode is available when you load it (with `dmesg`)? The sysfs files for the module attributes get created even if the calibration mode is unavailable.
* I also noticed that the health mode does not discharge the battery. I suppose that is expected. 

Do you happen to know the behavior of the Acer Care Center on your laptop regarding those two issues ?

R3G3N3R4T0R, 2022-09-27:
This is what I got from dmesg 
```
[111316.173284] acer_wmi_battery: loading out-of-tree module taints kernel.
[111316.173403] acer_wmi_battery: module verification failed: signature and/or required key missing - tainting kernel
[111316.188798] acer_wmi_battery: available modes: health mode, calibration mode
[111316.188802] acer_wmi_battery: active modes: health mode
```
So I guess it is there. It does have the charge and discharge cycle as I have said, but mine starts charging first instead of discharge.
I have not used Acer Care Center because I installed Linux on it immediately. I haven't had charging time estimation issues as it was quite accurate out of the box so I didn't really care.
I did try calibration mode before having a way to validate it though, so there is a chance it is already calibrated, I don't think it has that much precision to get the same exact result twice.

frederik-h, 2022-11-14:
> So I guess it is there. It does have the charge and discharge cycle as I have said, but mine starts charging first instead of discharge.
I said above that the calibration mode starts discharging directly for me. But this was with a fully charged battery. The battery will first be charged if is isn't at 100% yet. So the behavior you are seeing is expected.

closed 2022-11-14 by frederik-h