frederik-h/acer-wmi-battery #9: Not working on Aspire 5 A515-45 and a Swift 1 SF113-31

frederik-h, 2022-11-14:
@mnlipp has [reported](https://github.com/linrunner/TLP/issues/596#issuecomment-1312704984) the following issue: 

> I tried this on an Aspire 5 A515-45 and a Swift 1 SF113-31.
>
> echo 1 > /sys/bus/wmi/drivers/acer-wmi-battery/health_mode does not give an error, but
> cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode returns 0 (before and after the echo).

> So I suppose it doesn't work. But as this device has never seen Windows, therefore I cannot tell if the hardware is supposed to support > this (cannot check using "Acer Care Center").

@mnlipp: Could you please take a look at the dmesg output? There should be a line from the module which lists the available modes. Does it include the health mode?

mnlipp, 2022-11-14:
After booting the Aspire 5 A515-45 a few more times it seems that "active health mode" has been a coincidence. I couldn't reproduce this any more.

Logarithmus, 2023-02-20:
I have A515-45-R9FY and it doesn't work with `echo`, but does work by setting a kernel module option in `/etc/modprobe.d/acer_wmi_battery.conf` (create the file):
```modconf
options acer_wmi_battery enable_health_mode=1
```

Logarithmus, 2023-02-20:
@frederik-h although I have another problem: currently I have 95% battery charge, unplugged. When I plug the cable, it jumps to 100%. When I unplug, it becomes right again. Is it expected behaviour?

frederik-h, 2023-03-16:
> @frederik-h although I have another problem: currently I have 95% battery charge, unplugged. When I plug the cable, it jumps to 100%. When I unplug, it becomes right again. Is this expected behaviour?

That's not the behavior that I see on my laptop. However I can imagine that the different Acer models can behave differently in this regard.

closed 2023-05-13 by frederik-h