frederik-h/acer-wmi-battery #10: Swift X - SFX14-41G

brianmorgan805, 2022-12-25:
I have a Swift X - SFX14-41G and I can confirm that the health mode works and it stops charging at 80%. 

I've noticed that if I plug the charger in if the battery is >80% it doesn't stop charging to let the battery drop back to 80%. It stays at the battery level it was when it was plugged in.  I am planning to test the calibration mode so I will report back. 

`acer_wmi: Unknown function number - 8 - 0 #unplugged`
 `acer_wmi: Unknown function number - 8 - 1 #plugged in
`
This appeared in dmesg after activating health mode. I'm unsure if this is relevant to your driver or if it is an acer specific thing. I haven't noticed any issues and the driver appears to work as expected. Thanks so much.

art-tapin, 2022-12-31:
Hey, @freddy8051
First of all, thank you for testing on Swift X - SFX14-41G! I am using Fedora 37 and having the same model.

I am just curious,
1. What distro and kernel do you use? 
2. What is the experience by so far?
3. @frederik-h With the future kernel updates, will it be an issue if the native module will be added to the kernel supporting acer wmi batteries natively?

Thanks!

freddy8051, 2022-12-31:
@art-tapin 
I use EndeavourOS and I'm using the [6.1.1-AMD kernel](https://aur.archlinux.org/packages/linux-amd)  mainly because of the AMD p-state drivers being included prior it being mainlined and I haven't bothered checking if others work. I can try testing it on another kernel if needed.

the unknown function 8-0 and 8-1 appears in dmesg when the charger is plugged in (usb c or the barrel charger), but I imagine it's because of health mode. Outside of that I haven't noticed anything abnormal any my laptop has been functioning as expected.

art-tapin, 2022-12-31:
@freddy8051 thank you so much! Have you tried the similar module that was developed by [maxco2](https://github.com/maxco2/acer-battery-wmi) 

 Could you please try it on Fedora 37 Workstation 6.1 stock kernel?

freddy8051, 2023-01-07:
I'm not a fedora user, but maybe you can try it and let us know :) 

One thing i found is if I let my battery run out it doesn't save health mode unless the module is loaded during boot. i think this might be an acer limitation but it's an easy fix to have the module load. so far everything has worked great.

closed 2023-01-07 by freddy8051