frederik-h/acer-wmi-battery #14: Working on AN515-57-70MT but not updating sysfs files

asant, 2023-01-17:
Hi,
I'm reporting this module as working on Acer Aspire AN515-57 (model 70MT).
Both calibration and health modes seem to be working fine too, and so does changing their values. The only issue is that, unless I load the module by setting the enable_health_mode parameter, values in the files under the sysfs do not change.

For instance, say that health mode is disabled and I want to enable it:
$ cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
0
$ echo 1 | sudo tee /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
# entering password...
# after that, the charging light actually turns off, so it worked, but the value in sysfs is still 0
$ cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
0
# but if I reload the module, the value gets updated too
$ sudo rmmod acer-wmi-battery
$ sudo modprobe acer-wmi-battery
$ cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
1

David0601d, 2023-01-31:
Hello, I got the similar issue. 
$ cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
modprobe: FATAL: Module acer-wmi-battery not found in directory /lib/modules/5.15.0-58-generic
I type this command , but why got this error?

maniacx, 2023-03-13:
> Hi, I'm reporting this module as working on Acer Aspire AN515-57 (model 70MT). Both calibration and health modes seem to be working fine too, and so does changing their values. The only issue is [...]

Hey. According to your test health mode applies/write in sysfs but cannot be read back using sysfs.
How about calibration mode? Can you enable it using sysfs?
`echo 1 | sudo tee  /sys/bus/wmi/drivers/acer-wmi-battery/calibration_mode `
Also can it be read back using
`cat /sys/bus/wmi/drivers/acer-wmi-battery/calibration_mode`

If **calibration mode** status can be read back using sysfs , but  **health mode** cannot, it could be because update_state is not used in health mode store, but it used for calibration mode. 

Could you apply this patch and try it on your device ?https://github.com/maniacx/acer-wmi-battery/commit/4747abd90d1e8135a87302bd91177ab2431eec89

If **calibration mode** cannot be read by sysfs then dont try this patch, as it could be some other issue that is preventing from reading from sysfs.

I added acer device to my extension  [Battery Health Charging ](https://extensions.gnome.org/extension/5724/battery-health-charging/) a while back for changing threshold ,only to realize later , that  reading using using sysfs doesnt work and my extension depends on it. Would appreciate if you could test the patch, but at your own risk, as I dont have an acer device to test.

asant, 2023-03-13:
Hi there,
yes, I can confirm that changing the value inside calibration_mode works fine. The patch for health_mode works as well, thank you!

maniacx, 2023-03-13:
Hey asant. Are you using a gnome desktop? If yes could you also check if my extension works for it?
Battery Health Charging 
https://extensions.gnome.org/extension/5724/battery-health-charging/

asant, 2023-03-13:
> Hey asant. Are you using a gnome desktop? If yes could you also check if my extension works for it? Battery Health Charging https://extensions.gnome.org/extension/5724/battery-health-charging/

Sure thing, your extension works just fine on Fedora 37 and Gnome Shell 43.3.

maniacx, 2023-03-15:
Thanks for you help.

closed 2023-05-13 by frederik-h