frederik-h/acer-wmi-battery #24: Can support for AN515-45 running on another distro be added?

RexPacificusVII, 2023-06-04:
Getting this on Acer Nitro AN515-45 running on Pop!_OS 22.04 LTS

make -C /lib/modules/6.2.6-76060206-generic/build M=/home/fj/acer-wmi-battery modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.6-76060206-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
  You are using:           gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
  CC [M]  /home/fj/acer-wmi-battery/acer-wmi-battery.o
  MODPOST /home/fj/acer-wmi-battery/Module.symvers
  CC [M]  /home/fj/acer-wmi-battery/acer-wmi-battery.mod.o
  LD [M]  /home/fj/acer-wmi-battery/acer-wmi-battery.ko
  BTF [M] /home/fj/acer-wmi-battery/acer-wmi-battery.ko
Skipping BTF generation for /home/fj/acer-wmi-battery/acer-wmi-battery.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.2.6-76060206-generic'

frederik-h, 2023-06-05:
There is no special support needed. I think the module should still work. But see e.g. [here](https://askubuntu.com/questions/1348250/skipping-btf-generation-xxx-due-to-unavailability-of-vmlinux-on-ubuntu-21-04) for a workaround concerning for your BTF issue.

closed 2023-06-05 by frederik-h

RexPacificusVII, 2023-06-05:
Copied the vmlinux file to the correct directory using the workaround. However, still getting the following error.

make[1]: Entering directory '/usr/src/linux-headers-6.2.6-76060206-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
  You are using:           gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
make[1]: Leaving directory '/usr/src/linux-headers-6.2.6-76060206-generic'

frederik-h, 2023-06-05:
That's not related to this module.  Having said that, I think you can ignore the warning in this case. I get a similar warning and the module works fine for me.