frederik-h/acer-wmi-battery #25: How do you get WMI GUID of acer battery?

milynox, 2023-06-07:
I'm new to programming in Linux and really curious how you get WMI GUID of battery configuration. 

I try to do some researches and these leads me to Event Viewer in Windows. However I find it so hard & painful to locate this info in Event Viewer. 

Any other ways to try? @frederik-h

frederik-h, 2023-06-12:
You can get this information from the WMI MOF information that is contained in your ACPI DSDT. The tables can be dumped on Linux using acpidump. You can then decompile the MOF, e.g. using this [tool](https://github.com/pali/bmfdec). This contains the GUIDs. But this information alone does not tell you how to call the WMI functions. For this, I decompiled the Acer Care Center application which is based on .Net using [ilspy](https://github.com/icsharpcode/ILSpy).

milynox, 2023-06-19:
Wow, this is totally new to me. Must take a little bit of time to learn about these. Many thanks for your info @frederik-h

closed 2023-07-29 by frederik-h