frederik-h/acer-wmi-battery #27: ACER Aspire A315-42 battery limit 80% no reaction to the command

dima333750, 2023-07-08:
Hello everyone
I have an ACER Aspire A315-42, I really wanted to limit the battery because it is constantly on charge
Acted according to the instructions

```
sudo apt install build-essential linux-headers-$(uname -r) git
git clone https://github.com/frederik-h/acer-wmi-battery.git
cd acer-wmi-battery
make
sudo insmod acer-wmi-battery.ko

echo 1 | sudo tee /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
```

There is no reaction, the command just freezes, there is no error

I also tried
`sudo insmod acer-wmi-battery.ko enable_health_mode=1`
It shows no errors and the command passes, but the limit is not included

Per team 
```
cat /sys/bus/wmi/drivers/acer-wmi-battery/health_mode
-1
```

bios of the latest version, am I doing something wrong or does my laptop have no support?

frederik-h, 2023-07-29:
It seems that the laptop is not supported (cf. e.g. [this thread on the acer forum](https://community.acer.com/en/discussion/634746/set-battery-limit-threshold-for-aspire-a315-42-r0w-1-3). However, I don't think that the write should freeze in this case. Could you take a look at the kernel output with dmesg?

frederik-h, 2023-07-29:
See also issue  #23 which reports that another Aspire 315 model is not supported.

closed 2023-07-30 by frederik-h