frederik-h/acer-wmi-battery #28: How to make it persistent

Friptick, 2023-07-23:
Thanks for your work on this module.

It probably seems obvious to you but it would be really helpful if you could add a couple of lines to the instructions about how best to make the feature persistent.

For example, `insmod` will load the module on-the-fly, so it's not going to survive a reboot, right? And I notice (before attempting to load the module) that `echo 1 | sudo tee /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` refers to a non-existent path on my system, is that to be expected?

frederik-h, 2023-07-29:
The precise answer depends on your distribution. You should have a file like `/etc/modules` where you can add a line to indicate that the module should be loaded. You can pass `enable_health_mode=1` to the module to enable health mode. You would also have to put the module in the right place below `/lib/modules/$(uname -r)/kernel` to ensure that it will be found.

The sysfs file will only be created by the module.

Friptick, 2023-07-30:
Thanks. So on Debian-Ubuntu I now have this in `/etc/modules-load.d/acer-wmi-battery.conf`. I am trying to make this whole thing as portable as possible so I don't have to waste an hour working it all out again after an OS reinstall! Might be worth putting something like this in the README to help the less savvy Linux users benefit from your module.

```
## compile acer-wmi-battery.ko from https://github.com/frederik-h/acer-wmi-battery
#d="/lib/modules/$(uname -r)/kernel/drivers/custom/" && mkdir -p $d && cp acer-wmi-battery.ko $d && depmod
acer-wmi-battery enable_health_mode=1
```

Can be closed.

closed 2023-07-30 by frederik-h

Friptick, 2023-09-25:
Reopening because it broke! :)

After a routine OS upgrade the kernel was upgraded and the path `/lib/modules/$(uname -r)/kernel/drivers/custom/` obviously became outdated.

So I know Linux tech support is not your problem but I've done some research and can't see any obvious solution to make the module persistent other than waiting for it to break and then moving the `.ko` and re-running all the commands.

Am I'm missing something obvious?