frederik-h/acer-wmi-battery #34: Working on Aspire 5 (A517-52)

emperor06, 2023-09-13:
(coming from https://github.com/linrunner/TLP/issues/596)

Aspire A517-52, Archlinux 6.5.2
Installed as dkms module with Diman119's script from pull request.
The battery no longer sits on 100% charge (it's now at 80%).

```
# dmesg | grep -i acer
[    0.000000] DMI: Acer Aspire A517-52/Jasmine_TL, BIOS V1.34 04/21/2023
[    0.333292] integrity: Loaded X.509 cert 'Acer Database: 84f00f5841571abd2cc11a8c26d5c9c8d2b6b0b5'
[    9.494279] acer_wmi_battery: loading out-of-tree module taints kernel.
[    9.494282] acer_wmi_battery: module verification failed: signature and/or required key missing - tainting kernel
[    9.498315] acer_wmi_battery: available modes: health mode, calibration mode
[    9.498317] acer_wmi_battery: active modes: health mode
[   10.207714] input: Acer Wireless Radio Control as /devices/LNXSYSTM:00/10251229:00/input/input5
[   10.375958] acer_wmi: Acer Laptop ACPI-WMI Extras
[   10.375989] acer_wmi: Function bitmap for Communication Button: 0x801
[   10.386602] input: Acer WMI hotkeys as /devices/virtual/input/input9
```

```
tlp-stat -b
--- TLP 1.6.0 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = PANASONIC 
/sys/class/power_supply/BAT1/model_name                     = AP19B5L
/sys/class/power_supply/BAT1/cycle_count                    =     45
/sys/class/power_supply/BAT1/charge_full_design             =   3440 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   2806 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2258 [mAh]
/sys/class/power_supply/BAT1/current_now                    =      0 [mA]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT1/charge_control_end_threshold   = (not available) 

Charge                                                      =   80.5 [%]
Capacity                                                    =   81.6 [%]
```

frederik-h, 2024-03-21:
Thanks for your feedback!

closed 2024-03-21 by frederik-h

Shallrath, 2026-09-18:
The 80% charging limit seems to be persistent on this model. I loaded the module one single time to enable it and so far, it stayed enabled since 5 months.