frederik-h/acer-wmi-battery #36: DKMS to make it persistent?

Friptick, 2023-09-26:
Thanks for your module which works great.

Or at least it worked great until my OS (Ubuntu) did a routine upgrade and broke it. :(

After some research, it seems the fix for this is DKMS (Dynamic Kernel Module Support). Does that sound correct?

Any chance of a quick howto, or at least a warning of any obvious pitfalls, before I attempt implementing your module with DKMS and probably break everything? I know this is not your problem but some hints would be really appreciated, I've done some searching but surprisingly there's not much in the way of concrete examples of how to do this.

Friptick, 2023-09-28:
Oh!! Just noticed there is an outstanding pull request for exactly this! Duh. #31 

Please consider merging!

Thanks again for your work.

frederik-h, 2024-03-21:
> Oh!! Just noticed there is an outstanding pull request for exactly this! Duh. #31
> 
> Please consider merging!
> 
> Thanks again for your work.

It wasn't working when I last tried it. But it seems that it has been updated recently. I will take a look at it again.

closed 2024-03-21 by frederik-h