frederik-h/acer-wmi-battery #38: Working on Acer Nitro AN515-57 V1.19

brenobaptista, 2023-12-03:
But for some reason every week or so for no reason it stops working and I have to load the module again.

Not sure if it's because I power off my laptop and remove the power cable and only put it back the next day when powering on again. Maybe it gets below 80% because I remove it before completely powered off then for some reason it bypasses the module.

Should I maybe add the `sudo insmod acer-wmi-battery.ko enable_health_mode=1` command to run on system startup?

brenobaptista, 2023-12-15:
It fixed for a lot longer but after updating to new kernel a day later it stopped working again. I had to run `make` in the project again then I was able to use it.

[image]

frederik-h, 2024-03-21:
Thanks for the feedback. I'll take a look at the updated PR.

closed 2024-03-21 by frederik-h