frederik-h/acer-wmi-battery #39: Not working on Nobara Linux (Fedora-based) for kernel 6.6.9-200.fsync.fc39.x86_64

RexPacificusVII, 2024-01-04:
Getting this error after running `make`:

```
make -C /lib/modules/6.6.9-200.fsync.fc39.x86_64/build M=/home/rexpacificusvii/acer-wmi-battery modules
make[1]: Entering directory '/usr/src/kernels/6.6.9-200.fsync.fc39.x86_64'
  CC [M]  /home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.o
  MODPOST /home/rexpacificusvii/acer-wmi-battery/Module.symvers
  CC [M]  /home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.mod.o
  LD [M]  /home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.ko
  BTF [M] /home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.ko
/bin/sh: line 1: ./tools/bpf/resolve_btfids/resolve_btfids: No such file or directory
make[3]: *** [scripts/Makefile.modfinal:60: /home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.ko] Error 127
make[3]: *** Deleting file '/home/rexpacificusvii/acer-wmi-battery/acer-wmi-battery.ko'
make[2]: *** [/usr/src/kernels/6.6.9-200.fsync.fc39.x86_64/Makefile:1867: modules] Error 2
make[1]: *** [Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.6.9-200.fsync.fc39.x86_64'
make: *** [Makefile:5: all] Error 2
```

RexPacificusVII, 2024-01-04:
It was working with the previous kernel 6.6.8 without any issue

frederik-h, 2024-03-21:
This issue is unrelated to this module. I have googled for "resolve_btfids" and found several discussions about external module build issues. Perhaps you can find a fix in one of them.

closed 2024-03-21 by frederik-h