frederik-h/acer-wmi-battery #42: Working on Acer Nitro 5 (AN515-58)

autodistries, 2024-02-07:
Driver works with Acer Nitro 5 (AN515-58) on Linux NitroAN515-58 6.5.0-17-generic #17-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64 GNU/Linux with KDE Plasma.
While enabled, charging stops at 79%. Disabling immediately continues charge; re-enabling interrupts it again

autodistries, 2024-02-11:
Something weird happened. The driver was loaded and my PC plugged in, and it continuously charged to stay at 100%. 
After rebooting my linux, same issue. I booted into Windows then, and the Acer Care thing said that battery health mode was activated, but the PC was still charging up to 100%. I ran the battery calibration thing during the night, and now charging is indeed limited to 79% for my linux.
I would not know how to reproduce the issue. My best guess it that it needed to go below a certain battery percentage to effectively limit it.

frederik-h, 2024-02-12:
>  My best guess it that it needed to go below a certain battery percentage to effectively limit it.

Yes, that's true. You need to discharge to at most 79% for the limit to become effective.

closed 2024-02-12 by frederik-h