frederik-h/acer-wmi-battery #43: Input/Ouput Error with kernel 6.6.19-1 on Acer Aspire E15: Manjaro Linux

mcrose, 2024-03-10:
I installed current kernel linux66-headers (6.6.19-1) and base-devel

`╘⟹ make
make -C /lib/modules/6.6.19-1-MANJARO/build M=/run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery modules
make[1]: Entering directory '/usr/lib/modules/6.6.19-1-MANJARO/build'
  CC [M]  /run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery/acer-wmi-battery.o
  MODPOST /run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery/Module.symvers
  CC [M]  /run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery/acer-wmi-battery.mod.o
  LD [M]  /run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery/acer-wmi-battery.ko
  BTF [M] /run/media/mcrose/DATA/tmp.install/git/acer-wmi-battery/acer-wmi-battery.ko
make[1]: Leaving directory '/usr/lib/modules/6.6.19-1-MANJARO/build'
`
and when I try to run insmod:
`╘⟹ sudo insmod acer-wmi-battery.ko
insmod: ERROR: could not insert module acer-wmi-battery.ko: Input/output error
`

Acer Aspire E15 E5-575G-53VG
Manjaro Linux

frederik-h, 2024-03-21:
I suppose the WMI battery control interface is not supported on this model. Have you used the battery control from Windows before?

mcrose, 2024-03-21:
> I suppose the WMI battery control interface is not supported on this model. Have you used the battery control from Windows before?

Nop, I will try this weekend

closed 2025-03-16 by frederik-h