frederik-h/acer-wmi-battery #48: idk why am I getting this error... any solution?

Cortex0715, 2024-03-28:
cortex@cortex-Aspire-A715-42G:~/acer-wmi-battery$ sudo insmod acer-wmi-battery.ko
insmod: ERROR: could not insert module acer-wmi-battery.ko: Operation not permitted

MestoloPomello, 2024-03-30:
Do you have Secure Boot active? If you have and can't turn it off, you can try signing the module with a X.509 certificate and re-run the command. I did it some time ago by following the top answer [here](https://unix.stackexchange.com/questions/751517/insmod-causes-key-rejected-by-service), i don't know if there is a more straightforward way to do it

frederik-h, 2024-04-10:
Closing because this does not look like a problem that is specific to this module.

closed 2024-04-10 by frederik-h