frederik-h/acer-wmi-battery #52: Made an AUR Package of the DKMS Variant for Arch Linux

ryanbarillos, 2024-04-27:
Hi. 

I'd like to let you know that I've made an AUR package of the DKMS variant of this kernel, and I can say it's working 100% on my system and on my laptop. Hopefully this will get more exposure on the other side of Linux.

Link to AUR package: [here](https://aur.archlinux.org/acer-wmi-battery-dkms-git.git)
When I have the time again, I'll also work toward uploading the main variant to the AUR as well.

frederik-h, 2024-04-27:
Great!

ryanbarillos, 2024-05-03:
Done! There are now two ways to install this driver for Arch Linux: 

- [acer-wmi-battery-dkms](https://aur.archlinux.org/packages/acer-wmi-battery-dkms) (To get latest package release)
- [acer-wmi-battery-dkms-git](https://aur.archlinux.org/packages/acer-wmi-battery-dkms-got) (To get latest GitHub commit)

closed 2024-05-03 by ryanbarillos