frederik-h/acer-wmi-battery #53: Not working on an515-58 with nbfc-linux enabled

Bugaddr, 2024-04-27:


Bugaddr, 2024-05-05:
No specific errors are there in the dmesg as well as journalctl, i think NBFC-linux tools sets some registers values which enables fancontrol but will stop battery charging limits but i dont have idea of debugging that. I have attached my system details below

```
Operating System: Arch Linux                      
          Kernel: Linux 6.6.30-1-lts
    Architecture: x86-64
 Hardware Vendor: Acer
  Hardware Model: Nitro AN515-58
```

CarlosMolinesPastor, 2024-05-13:
Me too. I have the same problem and I install twice, normal and git package... When I can put 1 in /sys/bus/wmi/drivers/acer-wmi-battery/health_mode, grab this action but then when i plug it stop battery charging limits

CarlosMolinesPastor, 2024-05-13:
I have nitro 5 an515-57

Bugaddr, 2024-05-13:
@CarlosMolinesPastor can you please check @ryanbarillos suggestion ? My device is not currently not with me.

lovwanshichetan, 2024-07-01:
Same issue here on AN515-57

asant, 2024-10-15:
Hi there, I can confirm that this issue is there on AN515-57 too, but I managed to workaround it.
If you check the contents of the "Acer Nitro AN515-58.json" file of nbfc, you can see that it mentions setting the 0x03 register to either 17 or 81 to control the battery or not. So, I edited the "Acer Nitro AN515-57.json" to set that register to 81 and finally reloaded the configuration, and the problem was gone.

arch-holic, 2024-10-16:
> Hi there, I can confirm that this issue is there on AN515-57 too, but I managed to workaround it.
> If you check the contents of the "Acer Nitro AN515-58.json" file of nbfc, you can see that it [...]

Thanks a lot bud, i also have acer Nitro V same model AN515-57, i was scratching head when you found the solution maybe because i hadn't looked at other model's config file. Thanks once again, now i can use both nbfc & lgt lights together. Btw, are you facing constant heating in linux or is it due to thermal paste? My telegram: @unfiltered_me if so that we can discuss this further.

Bugaddr, 2024-10-21:
yeah setting it to 81 works.

arch-holic, 2025-01-23:
> yeah setting it to 81 works.

you can close this issue.

closed 2025-01-24 by Bugaddr