frederik-h/acer-wmi-battery #54: Not working on Acer Extensa 2519

dusxmt, 2024-05-18:
Hey there,

I just tried this kernel module, as I've replaced the battery on my laptop and want it to last longer than the previous one, and while this module does recognize the WMI battery interface, the battery status output buffer is 4 bytes on my machine instead of 8 bytes:

`[  163.603854] acer_wmi_battery: WMI battery status call returned a buffer of unexpected length 4`

I got this laptop back in 2017, so it's possible that it has an earlier version of the battery health interface.  I see that the 4 bytes at the end of the structure are function status bytes, with 0 being the health mode and 1 being the calibration mode.  I only have 0 available, so perhaps that means that only health mode adjustment is available

Is it reasonable to thing that it may be possible to get the module running with limited functionality, allowing it to configure just the health mode?

Thanks,
Marek

dusxmt, 2024-05-18:
Out of curiosity, I decided to dump what the WMI interface is returning for method 20:

ret.uFunctionList: 0x00, ret.uReturn[0,1] = { 0x00, 0x01 }, ret.uFunctionStatus[0] = 0x00

I tried to also call method 21 to set health mode, but it seems to not do anything, the data returned by method 20 are still the same.

I tried to see if I could extract some information about the interface from the DSDT, but unfortunately, there are no mentions of 79772EC5-04B1-4bfd-843C-61E7F77B6CC9 in the _WDG section or in the WDQQ BMF.  I also tried the fwts tool, and it sees that the interface exists, but it doesn't have a BMF:

```
FAILED [LOW] WMIMissingMethod: Test 1, GUID 79772EC6-04B1-4BFD-843C-61E7F77B6CC9 should have an associated method WMBF defined, however this does not seem to exist.
PASSED: Test 1, 79772EC6-04B1-4BFD-843C-61E7F77B6CC9 has more than zero instances
```

So unfortunately, it seems that this earlier version of the Acer Battery WMI interface has a different input/output argument format, and my ACPI tables don't contain any information about it.

frederik-h, 2024-05-21:
Thanks for sharing the results of your investigation!

closed 2024-05-21 by frederik-h