frederik-h/acer-wmi-battery #55: acer_wmi_battery: WMI battery status call returned a buffer of unexpected length 4

nemesazs, 2024-06-14:
Hi,
is this a not supported device?

having:
	acer travelmate p259-62-m
	mandjaro 6.6

both
	sudo insmod acer-wmi-battery.ko enable_health_mode=1
	sudo insmod acer-wmi-battery.ko

end up in:
	"insmod: ERROR: could not insert module acer-wmi-battery.ko: Input/output error"

sudo dmesg | tail -1

		"acer_wmi_battery: WMI battery status call returned a buffer of unexpected length 4"

Crypto-Darth, 2024-06-14:
Hello,

Did you try the charge limit function on windows? using the acer care centre?
Not sure if it's supported in your laptop or not.

Here's an similar issue (https://github.com/maxco2/acer-battery-wmi/issues/1) .
 here you can run the `sudo fwts wmi - ` command and see the outputs and match it to this pattern 79772EC5-04B1-4BFD-843C........ 
 if it matches then you might have the function else most probably not.
 
 also i hope you had installed the proper linux headers which is a pre-requirement for this.

nemesazs, 2024-06-14:
Hi,
thanks for the prompt response!

my SNID acer support does not offer 'acer care centre' only 'acer control centre'
- control centre with regards to battery only offers 'battery health check', and that passes ok

sudo fwts wmi -
"Results generated by fwts: Version V24.03.00 (2024-04-08 03:21:26).
...
\_SB_.PCI0.WMID._WDG (3 of 12)
  GUID: 79772EC5-04B1-4BFD-843C-61E7F77B6CC9
  WMI Method:
    Flags          : 0x02 (Method)
    Object ID      : BE
    Instance       : 0x01
PASSED: Test 1, 79772EC5-04B1-4BFD-843C-61E7F77B6CC9 has associated method \_SB_.PCI0.WMID.WMBE
PASSED: Test 1, 79772EC5-04B1-4BFD-843C-61E7F77B6CC9 has more than zero instances
...
23 passed, 2 failed, 0 warning, 0 aborted, 0 skipped, 0 info only.

Test Failure Summary
==================================================================================================================

Critical failures: NONE

High failures: NONE

Medium failures: 1
 wmi: GUID A9D77EF6-55E0-4706-B139-96ACE13F2269 is unknown to the kernel, a driver may need to be implemented for this GUID.

Low failures: 1
 wmi: GUID 79772EC6-04B1-4BFD-843C-61E7F77B6CC9 should have an associated method WMBF defined, however this does not seem to exist.

Other failures: NONE
Test           |Pass |Fail |Abort|Warn |Skip |Info |
---------------+-----+-----+-----+-----+-----+-----+
wmi            |   23|    2|     |     |     |     |
---------------+-----+-----+-----+-----+-----+-----+
Total:         |   23|    2|    0|    0|    0|    0|
---------------+-----+-----+-----+-----+-----+-----+"

cat /etc/lsb-release                                             ✔ 
"DISTRIB_ID="ManjaroLinux"
 DISTRIB_RELEASE="24.0.2" 
 DISTRIB_CODENAME="Wynsdey"
 DISTRIB_DESCRIPTION="Manjaro Linux""

hostnamectl
" Static hostname: nemes-travelmatep259g2m
       Icon name: computer-laptop
         Chassis: laptop 💻
      Machine ID: 666e2f347e1742d29680711c19a5bfc4
         Boot ID: 8254a2b3543b4d338e71c1941c9f16a2
Operating System: Manjaro Linux                   
          Kernel: Linux 6.6.32-1-MANJARO
    Architecture: x86-64
 Hardware Vendor: Acer
  Hardware Model: TravelMate P259-G2-M
Firmware Version: V1.51
   Firmware Date: Fri 2019-05-31
    Firmware Age: 5y 2w 1d"

pacman -Q linux66-headers
"linux66-headers 6.6.32-1"

Crypto-Darth, 2024-06-16:
Hey,
sorry for the late reply got busy somewhere

If the acer control centre doesn't show that option that probably means that your hardware doesn't support this feature :(

nemesazs, 2024-06-16:
Hi,
I suspected so :(
Thanks for your time!

Regards,
/Attila
<)

cryptodarth ***@***.***> ezt írta (időpont: 2024. jún. 16.,
V, 13:00):

closed 2025-03-16 by frederik-h