frederik-h/acer-wmi-battery #57: This also working in Acer Nitro 5 AN515-58

FosRexx, 2024-07-24:
That's it, the battery health function worked in my AN515-58, I haven't checked the calibration mode, so cannot say anything about that. But yeah it works on AN515-58. Thank you for this awesome project.

VSCast, 2024-09-24:
How did you make it work?

FosRexx, 2024-09-25:
> How did you make it work?

I am on Arch Linux, so I installed the [aur package](https://aur.archlinux.org/packages/acer-wmi-battery-dkms) then just enabled battery health mode using this command.
`echo 1 | sudo tee /sys/bus/wmi/drivers/acer-wmi-battery/health_mode`

I had to install the kernel header package explicitly but that's about it, the settings does reset after some new kernel upgrade so I had to again run that command afterwards.

frederik-h, 2025-03-16:
Better late than never: Thanks for the feedback!

closed 2025-03-16 by frederik-h