frederik-h/acer-wmi-battery #59: Working on Swift Go 14  SFG14-63-R6PU

matteganau, 2024-08-17:
Working on Acer Swift Go 14  SFG14-63-R6PU running [Fedora KDE Plasma Desktop](https://fedoraproject.org/spins/kde/).

For anyone else using Fedora the installation of the kernel-headers was slightly different:

`sudo dnf in kernel-devel dwarves`

I've started a calibration. If I wait until it starts to discharge and then leave it overnight will I know it has finished if it's charged to 100% again?

Thanks for creating and maintaining this project. I initially enabled the battery health control in Windows before installing Linux and was left with a bit of a conundrum about how to disable/enable it before finding this project. Great stuff!

frederik-h, 2025-03-16:
> Working on Acer Swift Go 14 SFG14-63-R6PU running [Fedora KDE Plasma Desktop](https://fedoraproject.org/spins/kde/).

Thanks for the feedback!

closed 2025-03-16 by frederik-h

clemisch, 2026-11-11:
Fyi this issue is linked in MODELS.md as a reference for Aspire E15 not working but it's clearly not about that?