frederik-h/acer-wmi-battery #70: Make on Nitro AN515-57 and Ubuntu 24.04 LTS

xianbroc, 2025-01-02:
Hello
It seems that acer-wmi-battery works well for ACER laptops and I wanted to try it. 
I have encountered a problem during make : 
$ sudo make 
make -C /lib/modules/6.8.0-51-generic/build M=/home/xianne_acer/nouveau_depot_git/acer-wmi-battery modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-6.8.0-51-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/acer-wmi-battery.o
  MODPOST /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/Module.symvers
  CC [M]  /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/acer-wmi-battery.mod.o
  LD [M]  /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/acer-wmi-battery.ko
  BTF [M] /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/acer-wmi-battery.ko
Skipping BTF generation for /home/xianne_acer/nouveau_depot_git/acer-wmi-battery/acer-wmi-battery.ko due to unavailability of vmlinux
make[1] : on quitte le répertoire « /usr/src/linux-headers-6.8.0-51-generic »
$ sudo insmod acer-wmi-battery.ko
insmod: ERROR: could not insert module acer-wmi-battery.ko: Key was rejected by service
Could I have some help ?

0x7375646F, 2025-01-04:
simple fix would be disabling the secure boot.

psiwray, 2025-03-28:
This answer https://unix.stackexchange.com/a/751571 works, tested it now.

xianbroc, 2025-03-29:
> Hello It seems that acer-wmi-battery works well for ACER laptops and I wanted to try it. I have encountered a problem during make : $ sudo make make -C /lib/modules/6.8.0-51-generic/build [...]

"[acer-wmi-battery]" was installed and worked for a while. But now I have an error that appears every hour in a box about something like "missing dependencies". 
I want to go back in a previous state where no "[acer-wmi-battery]" was on my computer. 
For an unknown reason, I have lost my /home/xianne_acer/nouveau_depot_git/acer-wmi-battery repositery. 
How can I remove all traces of  "[acer-wmi-battery]" ?
The running kernel is 6.8.0-56-generic
Thanks for any support.

xianbroc, 2025-03-29:
> simple fix would be disabling the secure boot.

"[acer-wmi-battery]" was installed and worked for a while. But now I have an error that appears every hour in a box about something like "missing dependencies". 
I want to go back in a previous state where no "[acer-wmi-battery]" was on my computer. 
For an unknown reason, I have lost my /home/xianne_acer/nouveau_depot_git/acer-wmi-battery repositery. 
How can I remove all traces of  "[acer-wmi-battery]" ?
The running kernel is 6.8.0-56-generic
Thanks for any support.

DistroAgnostic, 2025-04-17:
why messing with ubuntu, when you know about their shenanigans. Just use mint or fedora or anything not not so messy. Also, lts has it's issue. I remember one acer module wasn't working for me when using lts kernel at the end i had to remove the lts kernel to make my fans work(i mean control them).