frederik-h/acer-wmi-battery #72: Working on Acer Aspire 5 A515-45-R74N, even while off, and on dualboot.

guipro99, 2025-01-04:
Using Arch Linux 6.12.6-arch1-1. Project was downloaded directly by git clone. Something quite interesting, is that the battery lifespan mode worked even while my laptop is powered-off, but shenanigans happen while I'm on windows 10, (dual-boot) it recognizes the battery is at 79% but still gives an estimate for reaching 100%, but tools like hwinfo seem to break when trying to verify battery consumption or charge rate (displays 0, even tough it's clearly charging/discharging, percentage even changes).

frederik-h, 2025-03-16:
> [...] but shenanigans happen while I'm on windows 10, (dual-boot) it recognizes the battery is at 79% but still gives an estimate for reaching 100%, but tools like hwinfo seem to break when trying [...]

Does this  only happen if you enable the health mode using this driver on Linux or do you see the same behavior if you enable it using the Acer Control Center on Windows?

DistroAgnostic, 2025-04-24:
> Using Arch Linux 6.12.6-arch1-1. Project was downloaded directly by git clone. Something quite interesting, is that the battery lifespan mode worked even while my laptop is powered-off, but [...]

try turning off 80% limit switch from acer care centre app in windows.