frederik-h/acer-wmi-battery #82: health mode not working

athuljayaram, 2025-04-14:
sudo insmod acer-wmi-battery.ko enable_health_mode=1

it is still charging beyond 80%

DistroAgnostic, 2025-04-24:
> make make -C /lib/modules/6.11.0-17-generic/build M=/home/juno/Downloads/acer-wmi-battery modules make[1]: Entering directory '/usr/src/linux-headers-6.11.0-17-generic' warning: the compiler differs from the one used to build the kernel The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 You are using: gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 CC [M] /home/juno/Downloads/acer-wmi-battery/acer-wmi-battery.o MODPOST /home/juno/Downloads/acer-wmi-battery/Module.symvers CC [M] /home/juno/Downloads/acer-wmi-battery/acer-wmi-battery.mod.o LD [M] /home/juno/Downloads/acer-wmi-battery/acer-wmi-battery.ko BTF [M] /home/juno/Downloads/acer-wmi-battery/acer-wmi-battery.ko Skipping BTF generation for /home/juno/Downloads/acer-wmi-battery/acer-wmi-battery.ko due to unavailability of vmlinux make[1]: Leaving directory '/usr/src/linux-headers-6.11.0-17-generic'
> 
> sudo insmod acer-wmi-battery.ko enable_health_mode=1
> 
> AN515-43-R2M9
> 
> it is still charging beyond 80%

at least load module first `sudo insmod acer-wmi-battery.ko` and if issue persists then i believe it has to do with Ubuntu itself( aka highly modified debian), i saw other people on Ubuntu facing issue too.

frederik-h, 2025-04-24:
> make make -C /lib/modules/6.11.0-17-generic/build M=/home/juno/Downloads/acer-wmi-battery modules make[1]: Entering directory '/usr/src/linux-headers-6.11.0-17-generic' warning: the compiler differs [...]

Could you share the output of running `sudo dmesg  | grep acer_wmi_battery` after you loaded the module?