frederik-h/acer-wmi-battery #92: Works on Nitro ANV15-41 on Ubuntu 24.04.2

EDLLT, 2025-07-21:
Splendid work!

Oleg4260, 2025-07-23:
I have the same model, works on Arch Linux as well

Oleg4260, 2025-07-24:
> > I have the same model, works on Arch Linux as well
> 
> Hey, since you have the exact same model, may I ask if there were any issues you faced with this laptop running linux? (Considering that you have the same model, I believe that means you also have the exact same specs rtx 3050 w. a ryzen 7-7735HS processor)
> 
> One minor issue with mine is the internal microphone not working in linux which, to be frank, I don't care about anyway

Yes, I have the same specs.
The only current issue I still have is with suspend because it only supports shitty s2idle and no deep sleep.

I used to have the same internal microphone issue. The cause is that the acp6x (AMD Audio Coprocessor, the device to which the mic is connected) driver didn’t have a quirk for this model. Support for it was added to mainline on June 23 and is available since Linux 6.15.7 release.
I advice you to upgrade the kernel or manually patch the driver by editing [sound/soc/amd/yc/acp6x-mach.c](https://github.com/torvalds/linux/blob/master/sound/soc/amd/yc/acp6x-mach.c) and adding this entry to the quirk list:

{
	.driver_data = &acp6x_card,
	.matches = {
		DMI_MATCH(DMI_BOARD_VENDOR, "RB"),
		DMI_MATCH(DMI_PRODUCT_NAME, "Nitro ANV15-41"),
	}
},

I initially tried patching an older kernel, but couldn’t get it working - probably because I was doing something wrong, but when 6.15.7 was released I just updated my system and the microphone started working.

TL;DR: Install the latest kernel, try building your own with the patch above or just wait for new updates if you don't really care.

EDLLT, 2025-07-24:
Oh, I see. Thank you for the elaborate response man!

> The only current issue I still have is with suspend because it only supports shitty s2idle and no deep sleep.
Well damn, I learn something new about linux everyday. You're right, I'm only getting s2idle in /sys/power/mem_sleep
Do you believe there are ways out there to fix this or are we out of luck?

Oleg4260, 2025-07-24:
> Oh, I see. Thank you for the elaborate response man!
>
> Well damn, I learn something new about linux everyday. You're right, I'm only getting s2idle in /sys/power/mem_sleep
> Do you believe there are [...]
I already tried enabling S3 sleep (known as deep in Linux) in advanced BIOS settings (can be opened by holding Fn+Tab while pressing the power button, then F2 as always), there is an option to enable S3 instead of Modern Standby (s2idle), but when I tried selecting S3 and saving the settings, the laptop hangs on black screen while powering up, after hard resetting it comes back alive but with Modern Standby selected again, meaning that S3 is completely broken on this model and is unsupported.

That means fixing s2idle somehow is our only option, but I haven't found a solution yet, many people have troubles with Micro$oft's "Modern Standby" on Linux. In my case, when I try to put the laptop to sleep it wakes up immediately, this is what I see in dmesg then:
```

[  565.527503] r8169 0000:02:00.0 enp2s0: Link is Down
[  567.162659] PM: suspend entry (s2idle)
[  567.170680] Filesystems sync: 0.008 seconds
[  567.218955] Freezing user space processes
[  567.220532] Freezing user space processes completed (elapsed 0.001 seconds)
[  567.220535] OOM killer disabled.
[  567.220536] Freezing remaining freezable tasks
[  567.221590] Freezing remaining freezable tasks completed (elapsed 0.001 seconds)
[  567.221592] printk: Suspending console(s) (use no_console_suspend to debug)
[  567.313928] pcieport 0000:00:08.3: quirk: disabling D3cold for suspend
[  567.314448] ACPI: EC: interrupt blocked
[  567.726183] amd_pmc AMDI0007:00: Last suspend didn't reach deepest state
[  567.746461] ACPI: button: The lid device is not compliant to SW_LID.
[  568.147927] ACPI: EC: interrupt unblocked
```

EDLLT, 2025-07-25:
> there is an option to enable S3

I've entered into advanced bios, but I couldn't find the S3 option. Under what section is it and what's the option called?
I'm suspecting that perhaps my laptop shipped with a newer bios version where that option is completely removed. 

I'm also facing the exact same issue with the s2idle as the laptop wakes up immediately upon suspending. Didn't manage to fix that either unfortunately.
I've also spent a few hours trying to get deep sleep working through trying a few dozen approaches including patching the ACPI table, all of them failed...

Oleg4260, 2025-07-25:
Mine had 1.13 from the shop and I updated it to 1.17 because I hoped it would fix sleep (I installed Tiny11 for updating it and also to test if the microphone and modern standby works at all), also 1.18 has been released already. The settings entry is in AMD PBS/Power saving Configuration, but I don't think you should try it, you will just get black screen after saving the settings and exiting the bios. I heard that modern systems don't support S3 on hardware level so there's no chance probably. You can try updating your bios if you want, the best way is to install Windows 11 on a small partition and run the official utility under it, but we have slightly different models so your version might differ.

Oleg4260, 2025-07-25:
tbh i don't remember if i tried switching it on 1.17, i lost hope already

Oleg4260, 2025-07-25:
Speaking of my dmesg output, `ACPI: button: The lid device is not compliant to SW_LID.` doesn't appear when I close the laptop (in this case it falls into a loop of suspending and instantly waking up) and it probably means nothing. The amd_pmc message
`amd_pmc AMDI0007:00: Last suspend didn't reach deepest state` seems to be the main error, maybe we really should dig into fixing s2idle instead of trying to enable S3 which seems impossible, if you find anything, please tell me

EDLLT, 2025-07-26:
> The settings entry is in AMD PBS/Power saving Configuration, it's name is S3/Modern Standby support, but I don't think you should try it, you will just get black screen after saving the settings and exiting the bios.

Damn, I tried it and the laptop took its sweet time booting up before the acer logo showed up.
Upon booting it automatically reverted to modern standby. I am thankful that Acer was competent enough to program an automatic reversal option upon boot failure

Believe it or not, I just discovered that windows, technically, is considered to have a live bootable version(the installation setup environment) that I could utilize to run the bios update utility allowing me to skip installing windows completely.
So, I updated the bios to v1.18(latest), tried the S3 option again and got the same result. Looks like you're correct, it's impossible to get deep sleep working on this laptop at its current state

EDLLT, 2025-07-28:
This isn't relevant to this suspension issue, but here are some other "tips and tricks" to get the best experience from the nvidia driver that I learned along the way

The most stable nvidia driver version for suspension/resuming is version 550 as other versions seem to crash/corrupt applications that utilize the GPU upon resuming(eg: vlc, blender, etc).

In ``/etc/default/grub``, make sure to include this ``nvidia-drm.modeset=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1`` then run update grub

For browser hardware acceleration to be utilized properly, go to brave://flags (link differs based on your browser) and enable these options

[image]

EDLLT, 2025-07-28:
Actually, perhaps the nvidia is not the issue, looking at the logs, the problem really is from AMD.
Maybe the reason the chvt method worked is because of the correlation of nvidia and amd both being GPUs which made me think that the issue stemmed from nvidia, when in reality, amd was the one that was failing to suspend correctly

```
/etc/default/grub
GRUB_CMDLINE_LINUX_DEFAULT="nvidia-drm.modeset=1 nvidia.NVreg_PreserveVideoMemoryAllocations=1 amdgpu.runpm=0"

lspci -s 0000:00:01.1 -v # Seems to be causing a wakeup event
echo GPP0 | sudo tee /proc/acpi/wakeup
```

The main issue that's causing it to fail to suspend
```
Jul 28 20:59:09 HN /usr/libexec/gdm-x-session[2697]: (WW) AMDGPU(0): flip queue failed: Invalid argument
Jul 28 20:59:09 HN /usr/libexec/gdm-x-session[2697]: (WW) AMDGPU(0): Page flip failed: Invalid argument
```

I'll try to modify that custom sleep script and target it towards amd using chvt

EDLLT, 2025-07-28:
Lmfao, I figured it out. The fix keeps on getting simpler and simpler

It's that god damn keyboard backlight. It uses proprietary drivers or whatever that aren't exposed to linux, so that's why the system fails to correctly turn it off. I had noticed a correlation between the keyboard backlight timeout and suspension success, that delay was a lucky guess as it happened to wait for the keyboard backlight to turn off

To turn off the backlight manually, press FN + F11. 

s2idle suspend should work perfectly now

Oleg4260, 2025-07-28:
That would be hilarious, but unfortunately turning the keyboard backlight off doesn't help in my case, still instant wakeup and amd_pmc error

Oleg4260, 2025-07-28:
> > Looks like the second option causes a problem in my case. I put the script you provided in the place of the old one, and my dmesg now says this:
> 
> What is the status of these services?
> systemctl status nvidia-suspend.service nvidia-resume.service
> 
> If they're masked, try unmasking them and enabling them then see if you still get that error

Guess we don't need that anymore, but they weren't unmasked

Oleg4260, 2025-07-29:
Btw, if you had keyboard timeout enabled (that's default) explains why suspend didn't work for me by closing the lid or pressing "Sleep" at the login screen: I liked the way the keyboard is glowing so I disabled the timeout in the BIOS long before, and my keyboard backlight was on all the time, while yours was off when you didn't touch it.

EDLLT, 2025-07-29:
Ok, I posted the bug report here

https://bugs.launchpad.net/ubuntu/+source/linux-hwe-6.14/+bug/2118955

>  only appears when the lid is open, it doesn't really prevent it from suspending but maybe it should also be reported?
For now, I think it'd be best to make this bug report completely focused on the root of the issue. They'd be able to see that warning as it's present in the logs though

Oleg4260, 2025-08-22:
@EDLLT 
I thought I could talk to you directly, sent you a friend request on Discord some time ago but you didn't accept it, so writing an update here:
- The keyboard issue apparently was fixed in Linux 6.16, I first noticed it on 6.16.1-zen kernel, it now suspends correctly and handles keyboard backlight automatically.
- I found a better solution than acer-wmi-battery: https://github.com/PXDiv/Div-Acer-Manager-Max this is a great alternative to Nitrosense with power and fan controls, includes battery health features as well.